mikalhart/IridiumSBD #7: Needs extra overloading for 32-bit systems

majenkotech, 2016-08-31:
The `dbg()` and `console()` overloads used for `size_t` don't exist on 32-bit systems. On 8-bit systems `size_t` is the same as `uint16_t` which gets covered by the `uint16_t` overload of the functions. On 32-bit systems it's the same as `uint32_t` and there is no overloading for that variable type.

````C++
void IridiumSBD::dbg(uint32_t n)
{
#if ISBD_DIAGS
   if (this->pDiagsStream)
      pDiagsStream->print(n);
#endif
}

void IridiumSBD::console(uint32_t n)
{
   if (this->pConsoleStream)
      pConsoleStream->print(n);
}
````

majenkotech, 2016-08-31:
Half could be fixed with PR #5, but adding the proper overloading is a better long-term solution.

mikalhart, 2017-10-21:
Thank you @majenkotech.  This is fixed in the new 2.0 beta.

closed 2017-10-21 by mikalhart