mikalhart/IridiumSBD #9: esp8266 crashing

jongross, 2018-01-03:
Here is my code (platformio):

```
#include <Arduino.h>
#include <SoftwareSerial.h>
#include "IridiumSBD.h"

/* receive / transmit / invert logic / buffersize*/
SoftwareSerial swSer(14, 12, false, 256);

// Declare the IridiumSBD object
IridiumSBD modem(swSer);

void setup()
{
	Serial.begin(19200);
	Serial.println("Starting setup...");

	int signalQuality = -1;
        int err;

  // Start the serial port connected to the satellite modem
	Serial.println("Starting satellite serial connection (swSer)");
	swSer.begin(19200);

  // Begin satellite modem operation
  Serial.println("Starting modem...");
  err = modem.begin();
	Serial.print("modem: ");
	Serial.println(err);

  if (err != ISBD_SUCCESS)
  {
    Serial.print("Begin failed: error ");
    Serial.println(err);
    if (err == ISBD_NO_MODEM_DETECTED)
      Serial.println("No modem detected: check wiring.");
    return;
  }
}

void loop() {
}

```
A simple test to see if the library runs - nothing wired to the esp (using adafruit huzzah), esp is known good, but when I run this code on it, I get constant reboots after calling modem.begin().

Sample output: 

```
X�␘~�H�Starting setup...
Starting satellite serial connection (swSer)
Starting modem...
�X
X�B�
X�␘~�(�Starting setup...
Starting satellite serial connection (swSer)
Starting modem...
FX
X�B�
X�␘~�"�Starting setup...
Starting satellite serial connection (swSer)
Starting modem...
F8

X�B�
X�␘~�H�Starting setup...
Starting satellite serial connection (swSer)
Starting modem...
```

Expected output would be a detection that there is no modem detected, I would think, instead of rebooting.

Can you help me understand what I'm doing wrong?

mikalhart, 2018-01-03:
Jon, could you define ISBDConsoleCallback and ISBDDiagsCallback, per the example code?  modem.begin() is a rather lengthy routine, and it's hard to guess what might cause a reset.  These will help.

jongross, 2018-01-03:
You bet, new output:
```
X�
X�␘~�H�4E��ѥ���������rrj�4E��ѥ�����ѕ���ѕ���ɥ���␚�����ѥ���B��M�ɥjR4E��ѥ���j�����rrj�4*��+��␖R-�W�.+␖KT�V-�H�>
> AT
Waiting for response OK

<<
Soft WDT reset

ctx: cont
sp: 3ffef390 end: 3ffef620 offset: 01b0

>>>stack>>>
3ffef540:  00000000 00000002 3ffee548 40202944
3ffef550:  00000001 00000002 3ffee548 40202abb
3ffef560:  00000000 00000000 00000000 3ffe89c5
3ffef570:  0000033d 00000002 402312d5 40203279
3ffef580:  00000001 402312d2 3ffee548 3ffee5f0
3ffef590:  0000033c 3ffe89c5 3ffee548 40202d51
3ffef5a0:  3fffdad0 00000011 3ffee5cc 40201225
3ffef5b0:  3ffe89c7 00004b00 3ffee5cc 40201225
3ffef5c0:  3ffe8467 feefeffe 3ffee5cc 3ffee5f0
3ffef5d0:  3ffee548 00000000 3ffee548 40202d90
3ffef5e0:  3ffee548 00004b00 3ffee5cc 40203440
3ffef5f0:  feefeffe feefeffe feefeffe ffffffff
3ffef600:  3fffdad0 00000000 3ffee5e8 40201660
3ffef610:  feefeffe feefeffe 3ffee600 40202648
<<<stack<<<
```

jongross, 2018-01-16:
Hey @mikalhart - anything I else I can provide to assist with troubleshooting?

vangalvin, 2019-09-24:
Did you manage to resolve this? it looks like the ESP32 has a similar issue,

mikalhart, 2019-10-13:
Please try new release 2.0.0 and see if this is resolved.  There was some misunderstanding about how the "weak" attribute works, and this has been changed.