mikalhart/IridiumSBD #10: Feature request: Clear MO/MT buffers using AT+SBDD

PaulZC, 2018-02-04:
Hi Mikal,
Not an issue as such - more a feature request for the next release:
Could you add a function to clear the MO/MT/both buffers using AT+SBDD0/1/2 ?
The reason I ask is that:
If I send an MO message using sendSBDText
then send an empty message to check for an MT reply using sendReceiveSBDText(NULL, sbdBuffer, bufferSize)
the 9603N automatically resends the old text message stored in the MO buffer!
I'm using a messy workaround in my code to force the clear using ssIridium.println("AT+SBDD0"); and then discarding the reply.
It would be great to have this incorporated properly into the library.
Sincere thanks,
Paul

mikalhart, 2019-10-13:
This is a good idea, but it's not implemented yet.  Keeping it alive for future work.

PaulZC, 2019-10-14:
Thanks Mikal.
I did put a solution together a while ago:

In the .h:

```
#define ISBD_CLEAR_MO		0
#define ISBD_CLEAR_MT		1
#define ISBD_CLEAR_BOTH		2
//public:
   int clearBuffers(int buffers = ISBD_CLEAR_MO);
//private:
   int internalClearBuffers(int buffers = 0);
```

In the .cpp:

```
//public:
// High-level wrapper for AT+SBDD
int IridiumSBD::clearBuffers(int buffers)
{
   if (this->reentrant)
      return ISBD_REENTRANT;

   this->reentrant = true;
   int ret = internalClearBuffers(buffers);
   this->reentrant = false;
   return ret;
}
//private:
int IridiumSBD::internalClearBuffers(int buffers)
// Clear the MO/MT/Both buffers
// Defaults to clearing the MO buffer to avoid resending old messages
{
   if (this->asleep)
      return ISBD_IS_ASLEEP;

   if (buffers == ISBD_CLEAR_MT) // Clear MT buffer
   {
      send(F("AT+SBDD1\r"));
   }
   else if (buffers == ISBD_CLEAR_BOTH) // Clear both buffers
   {
      send(F("AT+SBDD2\r"));
   }
   else // Clear MO buffer
   {
      send(F("AT+SBDD0\r"));
   }
   if (!waitForATResponse())
      return cancelled() ? ISBD_CANCELLED : ISBD_PROTOCOL_ERROR;

   return ISBD_SUCCESS;
}
```

If you would like a Pull Request, just let me know.
All the best,
Paul