mikalhart/IridiumSBD #16: Case Sensitivity Conflicts between time.h & Time.h

adamgarbo, 2018-05-07:
Hi there,

I discovered today, when migrating my code over to version 2.0 of the IridiumSBD library, that if you have [Paul Stoffregen's Time library](https://github.com/PaulStoffregen/Time) installed, you may encounter problems when attempting to compile your code while using case sensitive filesystems like macOS. 

This appears to be due to the fact the Paul has a `Time.h` file included in his library that conflicts with  the POSIX time.h library included by IridiumSBD.cpp. It seems users of the RTCZero library have [also experienced this problem](https://github.com/arduino-libraries/RTCZero/issues/28). Since Paul will be keeping this file [to preserve compatibility with older versions of Arduino](https://github.com/PaulStoffregen/Time/issues/74), the solution appears to be to simply delete the `Time.h` file from your `/libraries/Time/` folder.

I'm wondering if it may be worthwhile to implement some form of check that the IridiumSBD library could perform to ensure it is compiling with the correct `time.h`?

For reference, the error messages I received when attempting to compile with Paul's `Time.h` file present in my library are:

```
/Users/adam/Documents/Arduino/libraries/IridiumSBD-2.0/src/IridiumSBD.cpp: In member function 'int IridiumSBD::getSystemTime(tm&)':
/Users/adam/Documents/Arduino/libraries/IridiumSBD-2.0/src/IridiumSBD.cpp:203:14: error: aggregate 'tm epoch_start' has incomplete type and cannot be defined
    struct tm epoch_start;
              ^
/Users/adam/Documents/Arduino/libraries/IridiumSBD-2.0/src/IridiumSBD.cpp:223:43: error: 'mktime' was not declared in this scope
    time_t epoch_time = mktime(&epoch_start);
                                           ^
/Users/adam/Documents/Arduino/libraries/IridiumSBD-2.0/src/IridiumSBD.cpp:225:30: error: 'localtime' was not declared in this scope
    memcpy(&tm, localtime(&now), sizeof tm);
                              ^
/Users/adam/Documents/Arduino/libraries/IridiumSBD-2.0/src/IridiumSBD.cpp:225:40: error: invalid application of 'sizeof' to incomplete type 'tm'
    memcpy(&tm, localtime(&now), sizeof tm);
                                        ^
```

Cheers,
Adam

jacky4566, 2018-12-20:
Same Issue here. Thanks for the tip!

I had to remove Time.h from both locations 
```
\libraries\Time\  
\hardware\teensy\avr\libraries\Time
```

mikalhart, 2019-01-24:
@maxholgasson,

I think that implies you are _using_ the Time library, right?  I _think_ the solution proffered here--deleting the header--only works to remove the conflicts it causes in other libraries.

All... correct me if I have misunderstood.

adamgarbo, 2020-07-15:
Closed with the accepted solution to simply delete or rename the Time.h file in Paul Stoffregen's [Time](https://github.com/PaulStoffregen/Time) library.

I would also direct new users to the official [SparkFun IridiumSBD library](https://github.com/sparkfun/SparkFun_IridiumSBD_I2C_Arduino_Library), which is an actively maintained fork of Mikal's IridiumSBD library, modified by [@PaulZC](https://github.com/PaulZC/).

closed 2020-07-15 by adamgarbo