mikalhart/IridiumSBD #18: GCC compiler error when trying to compile the example sketch 'PassThru'

Chase0fSpades, 2018-10-09:
Arduino: 1.8.6 (Windows 10), TD: 1.44, Board: "Arduino/Genuino Uno"

lto1.exe: internal compiler error: in lto_output_varpool_node, at lto-cgraph.c:624

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

per1234, 2018-10-09:
This problem only occurs with the avr-gcc 5.4.0-atmel3.6.1-arduino2 used by Arduino AVR Boards 1.6.22 and newer. The relevant bug report is here:
https://github.com/arduino/ArduinoCore-avr/issues/39
A code workaround is described in that thread.

---
An easier, though less desirable, workaround is to roll back Arduino AVR Boards to 1.6.21:
Tools > Board > Boards Manager
Wait for downloads to finish.
Click on "Arduino AVR Boards".
Select 1.6.21 from the dropdown menu.
Click "Install".
Wait for installation to finish.
Click "Close".

Due to a [bug](https://github.com/arduino/Arduino/issues/7973), you can not do this with Arduino IDE 1.8.6 but you can with any other Arduino IDE version.

If you have **File > Check for updates on startup** enabled, the Arduino IDE will periodically prompt you to update Arduino AVR Boards. You must refrain from doing so, otherwise the error will come back.

stkeane1, 2018-10-20:
So I had a thread about this issue here: https://forum.arduino.cc/index.php?topic=566280.0

internalBegin() causes the issue here

`// The usual initialization sequence
   FlashString strings[3] = { F("ATE1\r"), F("AT&D0\r"), F("AT&K0\r") };
   for (int i=0; i<3; ++i)
   {
      send(strings[i]); 
      if (!waitForATResponse())
         return cancelled() ? ISBD_CANCELLED : ISBD_PROTOCOL_ERROR;
   } 

`
The solution we found was 

`  // The usual initialization sequence
   const FlashString a = F("ATE1\r");

   const FlashString b = F("AT&D0\r");
   const FlashString c = F("AT&K0\r");
   const FlashString strings[3] = { a, b, c };
   for (int i=0; i<3; ++i)
   {
      send(strings[i]); 
      if (!waitForATResponse())
        return cancelled() ? ISBD_CANCELLED : ISBD_PROTOCOL_ERROR;
   }`

Hope this helps.

mikalhart, 2019-10-13:
Outstanding analysis @stkeane1.  Sorry for the abysmally slow reply.  I'll incorporate a variant of this in a forthcoming update.  Thanks again.

mikalhart, 2019-10-13:
Please try new release 2.0.0 and see if this is resolved.  There was some misunderstanding about how the "weak" attribute works, and this has been changed.