mikalhart/IridiumSBD #19: Protocol Error: From waitForATResponse().

stkeane1, 2018-11-07:
I have developed code using 5V pro trinkets that was able to send and receive data. I switched over to 3V Trinkets and am using the 3V and ground pin on the RockBlock. 

Switching to the 3V trinket has caused my modem to give me a protocol error when trying to send messages.

I have no problem turning on the modem and getting signal quality just sending a message. I am in a spot where I have been able to send messages in the past.

Upon looking into the library further I notice that waitForATResponse() is being returned false giving me the protocol error. Im not exactly sure why this function is being returned false, but it only started happening when I switched to the 3 volt trinket. I have tried the examples and they give the same protocol error when trying to sending a message. 

I will test with another 5v microprocessor when one becomes avaliable. Has anyone else had similar issues?

mikalhart, 2018-11-08:
@stkeane1, I think the RockBlock requires 5V for its power source.  Even though the Iridium module itself is 3.3V, the RockBlock requires 5V, which is regulated down to 3.3V for the module.

stkeane1, 2018-11-08:
@mikalhart Thanks for the response. Your library has done wonders for us in Arizona monitoring environmental conditions. 

This seems a bit odd considering they have a 3.7 volt lithium ion battery power pin. Possibly the 3.3 volts provided by the trinket is not enough for this pin. 

I adjusted the AT timeout to stop the protocol error, and allow the message to get sent but instead it just gives me MO codes: 18,32,38 (not the protocol error). So it appears it cannot make connection. Interesting enough it is able to get the last know signal quality. 

If 5 Volts is required I will look into boost converters because we really want to switch over to a feather wing products at 3.3 volts. Does this sounds practical? 

Again I appreciate all the time you put into this library, and any extra assistance you may provide.

Thank you

mikalhart, 2018-11-08:
@stkeane1, thanks for testing with the Trinket.  Good to know that it works... at least at 5V.
http://www.rock7mobile.com/downloads/RockBLOCK-Developer-Guide-Mk2.pdf
My memory of discussions with Rock7 is that if you power via the "LiIon" pin, you must be able to provide considerable current (7.5W).  If you are powering via the Trinket 3.3V line, you probably aren't able to.  Powering the RockBlock from the "5vIn" line reduced this requirement to only 450mA, but then, it must be 5V.

adamgarbo, 2018-11-10:
Rock Seven's new RockBLOCK 9603 modems now also have the 3.7V power supply pin limited to 450 mA. The Adafruit Trinket's 3.3V regulator would definitely would not be able to supply enough current for any older versions of the modem at this voltage.

stkeane1, 2018-11-13:
Thanks for letting me know. I will be testing a booster between the 3.3 Volt trinket and the RockBlocks 5V pin and I will let you know the results.

mikalhart, 2019-10-13:
@stkeane1, recent experiments comparing the older RB's to the new 9603 confirm our suspicion that the 9603 regulator is superior.  Did you get the old RB to work with your own 5V boost regulator?