mikalhart/IridiumSBD #23: ESP32 + RockBLOCK

vangalvin, 2019-09-24:
Has anyone managed to get the RockBLOCK working on an ESP32? 
Have tried several variants of the ESP32 and all seem to crash.

vangalvin, 2019-09-24:
I seem to be getting the following error.
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)

Just as the setup hits this line.
// Setup the Iridium modem
  modem.setPowerProfile(IridiumSBD::USB_POWER_PROFILE);

vangalvin, 2019-09-24:
Here is the full error I am getting, I have checked to make sure the unit is working and all seems to be OK. however when using the library I am now getting this.

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060f30  A0      : 0x800d11ce  A1      : 0x3ffb1ef0  
A2      : 0x3ffbfcf8  A3      : 0x3f401044  A4      : 0x3ffb8630  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0e74  A9      : 0x3ffb1e90  
A10     : 0x3ffbfcf8  A11     : 0x00000043  A12     : 0x3ffb8408  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb1ef0 0x400d11cb:0x3ffb1f10 0x400d13ae:0x3ffb1f50 0x400d0d41:0x3ffb1f70 0x400d225f:0x3ffb1fb0 0x40089fb9:0x3ffb1fd0

Rebooting...

mikalhart, 2019-10-13:
Please try new release 2.0.0 and see if this is resolved.  There was some misunderstanding about how the "weak" attribute works, and this has been fixed.

vangalvin, 2019-10-14:
Hey @mikalhart, I gave 2.0.0 a try and have the same issue. the addition of this in to the example fixed the problem.

`#if DIAGNOSTICS
void ISBDConsoleCallback(IridiumSBD *device, char c){
Serial.write(c);
}
void ISBDDiagsCallback(IridiumSBD *device, char c){
Serial.write(c);
}
#else
void ISBDConsoleCallback(IridiumSBD *device, char c){
// do nothing
}
void ISBDDiagsCallback(IridiumSBD *device, char c){
// do Nothing
}
#endif`