mikalhart/IridiumSBD #26: Modem doesn't want to start

infinitemind2, 2019-10-10:
Modem was starting, disconnected the device, now the code wont go pass the begin statement
with the follow output

Starting Modem -> Starting modem library... Calling internalBegin
Powering on modem...
..>> AT
Waiting for response OK

<< ..AT
OK
..>> ATE1
Waiting for response OK

<< ..............................................................Powering off modem...

	**** Modem failed: error -> Protocol Error
where should i look!

mikalhart, 2019-10-13:
@infinitemind2, what version of RB do you have?  Some early versions of RB were susceptible to a regulator failure that would manifest in this exact behavior.

Korka16, 2021-03-21:
Hi 
I have the same problem. i am using an arduino uno. could you find out why is it not working? Can you help me to understand how to fix it please