mikalhart/IridiumSBD #28: ESP32 + Rockblock

vangalvin, 2019-10-14:
Just a heads up if anyone is using an ESP32 and having issues with the examples. Change the DIAGNOSTICS lines to this and it all should work. 

`#if DIAGNOSTICS
void ISBDConsoleCallback(IridiumSBD *device, char c){
  Serial.write(c);
}
void ISBDDiagsCallback(IridiumSBD *device, char c){
  Serial.write(c);
}
#else
void ISBDConsoleCallback(IridiumSBD *device, char c){
 // do nothing
}
void ISBDDiagsCallback(IridiumSBD *device, char c){
// do Nothing
}
#endif
`

closed 2019-10-14 by vangalvin

mikalhart, 2019-10-14:
Hi, yes, thanks for this tip, @vangalvin.  However, I'm hoping that version 2.0.0 (just posted today) will fix this problem in the driver without the user having to modify any of the sample code.

vangalvin, 2019-10-14:
Awesome, I will give 2.0.0 a go as soon as i get the chance. I was just looking at the system and trying to work out if there was a way to just send a ring alert without sending a message via the API. sadly there is not. :(