mikalhart/IridiumSBD #30: License

JustinShetty, 2019-10-19:
Hi there,

I would like to use this library for an open-source project... did you exclude a license file on purpose? As-is, it's not legally advisable for anyone to use this library.

May I suggest GPLv2 or LGPL?

adamgarbo, 2019-10-19:
Hi, 

If you read either of the IridiumSBD.cpp or IridiumSBD.h src files, you'll find information on the license:

> This library is free software; you can redistribute it and/or
modify it under the terms of the GNU Lesser General Public
License as published by the Free Software Foundation; either
version 2.1 of the License, or (at your option) any later version.

Mikal's [website](http://arduiniana.org/libraries/iridiumsbd/), also states: 

> This library is distributed under the terms of the GNU LGPL license.

Cheers,
Adam

JustinShetty, 2019-10-20:
Can't believe I missed that, thanks for pointing it out. I think I was fixated on the lack of an explicit license file.

closed 2019-10-20 by JustinShetty