mikalhart/IridiumSBD #33: P

vestwind, 2020-06-29:


vestwind, 2020-06-29:
Hello, I am trying to use the BasicSend for sending a message to a RockBLOCK+ unit from an Arduino Mega 2560. 
I have tried with several pins, such as Serial2 and Serial3 as well as using software serial on pins 10, 11, 12 and 13.

All i get is: 

Starting modem... 
Calling internal begin
Powering on modem... 
>>AT 
Waiting for response OK

<< >> ATT
Waiting for response OK

I am supplying the RB+ with 12V, and i can see it oscillating between 0 mA and 219mA, every second or so. 
I have tried to change up the RX and TX, as mentioned in earlier issues. I have the Sleep Pin set to Pin 4. 
At this point I am out of ideas. I am quite inexperienced, so I have probably overlooked something. Please help.

LgArcida, 2020-10-01:
Any update on this issue?
I am facing the same problem