mikalhart/IridiumSBD #41: Use of Sleep pin causes Ring pin to not work correctly.

iliabaranov, 2022-08-29:


iliabaranov, 2022-08-29:
Seem to be able to correct this by adding this to the end of the setup section:
`modem.hasRingAsserted();`

It seems that using both sleep and ring causes ring to falsely go low during setup, reading it back has the effect of clearing this out

adamgarbo, 2022-08-29:
I don't believe this library is maintained any longer. 

I would recommend using the SparkFun fork maintained and enhanced by Paul Clark: https://github.com/sparkfun/SparkFun_IridiumSBD_I2C_Arduino_Library

PaulZC, 2022-08-29:
Hi Ilia (@iliabaranov ),

I ran into a similar ring pin issue a few years ago - possibly the same one. My fix - at the time - was the following:

"Included a fix to let the serial Ring example work properly (**hasRingAsserted** now checks the ringPin itself instead of assuming **cancelled** will be able to do it)."

See the changes to ```IridiumSBD.cpp``` in this commit:

https://github.com/PaulZC/IridiumSBD/commit/7a49765925e4798fc89fcc12fcae53a1bd57e4a2

The changes were included when we created the SparkFun fork. The SparkFun fork isn't intended to gatecrash Mikal's library, we gratefully took what he'd written and then added on I2C support for the "Qwiic Iridium" product.

I hope this helps,
Paul

adamgarbo, 2022-08-29:
Also to clarify, I only make the suggestion because I know there have been a number of issues that have been addressed in the SparkFun fork! 

@mikalhart do you think it could be worth making a note on your repository stating the library's status (e.g., active, inactive, etc.)? This could help give new and existing users a heads up as to the likelihood of their issues being addressed.

Cheers,
Adam

mikalhart, 2022-09-04:
Hi guys.  @PaulZC... thrilled to see that Sparkfun is supporting this great technology.  Yes indeed, I'm particularly glad that this issue with the apparent Ring/Sleep interdependence at least has a workaround.  (I thought it was something peculiar with my own setup!)

@adamgarbo I may do exactly what you suggest.  I'm traveling a lot these days and am not in a great position to do the testing necessary to support this work. 

Thanks, all, for comments.